usdot-fhwa-stol/carma-platform #776: CARMA Basic Travel UI - Initial Implementation - Layout

dan-du-car, 2020-07-27:
### Types of Issue

The developer needs to create a stand alone branch from integration/routing branch.
Integration/routing branch may have build issues therefore need to work with team members to resolve any issues that affects this user story. If not doable within the timeframe, document the issues in Github to be resolved and prioritized.
Login and Logout should be working order, including the desktop icon link, similar to CARMA Messenger layout.
The layout should cover the header, footer and display panels.
Route selection and display should be working.
Plugin display and selection should be working which includes displaying required plugins. Note will only assume what already has been implemented on Guidance.
This assumes that some of the topics, services, plugins and other capabilities may not be implemented or working order. Therefore developer could choose to dummy the data or have a placeholder at this time to display the new layout with the graphics and icons, which means some actual functionality may not work at the time.

UI look and feel changes according to UI mock design
project folder structure modifications
Reimplemented ros connection and ros topics

- [ ] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

### Expected behavior

### Software version this applies to

### Actual behavior

### Steps to reproduce the behavior

1. Do this
1. Then do this...

### Related work

Link to related tickets or prior related work here.

maefromm, 2020-08-13:
Reported the https://github.com/usdot-fhwa-stol/carma-platform/issues/761 closed that was found under this.

dan-du-car, 2020-09-01:
New PR is created : https://github.com/usdot-fhwa-stol/carma-web-ui/pull/47

dan-du-car, 2020-09-01:
https://github.com/usdot-fhwa-stol/carma-web-ui/pull/47

msmcconnell, 2021-03-01:
Resolved by above PRs

closed 2021-03-01 by msmcconnell