usdot-fhwa-stol/carma-platform #784: Windows Shared Folder appears to take up local storage space on Pacifica

msmcconnell, 2020-07-29:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
On the Pacifica the WindowsSharedFolder is being treated as part of the local drive space. This prevents additional logs from being generated. 
[image]

This occurs even while the NAS is running and appears to be functional, though I do not have the knowledge on how to validate the NAS is accepting data. 
### Expected behavior
CARMA can log data to the NAS in excess of the Spectra PC disk drive storage capacity. 
### Software version this applies to

develop after 3.3.0
### Actual behavior
See above picture
### Steps to reproduce the behavior
Run carma on vehicle and observe reported storage space on main Ubuntu drive. 

### Related work

Link to related tickets or prior related work here.

TonysCousin, 2020-07-29:
Sounds like the NAS sees this as a true shared folder (where both systems see all the contents as if they are local) rather than using the Ubuntu directory simply as a mount point.

jtbaird, 2020-09-02:
Resolved as part of isue #788. After rebuilding the hard drives, reinstalling Windows, and mounting the WindowsSharedFolder again, the WindowsSharedFolder showed up as a separate drive in the Spectra's disk usage analyzer, and did not appear to be taking up storage on the spectra

closed 2020-09-02 by jtbaird