usdot-fhwa-stol/carma-platform #785: CARMA cannot run while logging to NAS on Silver Truck

msmcconnell, 2020-07-29:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
While attempting to run a develop build of carma on the silver truck it was discovered that the NAS logging was preventing proper operation. ```carma start all``` would result in a hung launch which would also make the /opt/carma folder nearly unusable. It would take multiple minutes to run a basic ```ls``` command in that directory after running carma. It is not clear what the root cause is, but you can get around this issue by removing the symlinks to the shared drive folder. 
### Expected behavior
carma can run on the trucks from the develop branch
### Software version this applies to

develop after 3.3.0
### Actual behavior
carma start all results in a hanging launch that will eventually just fail and exit after afew minutes.
```
Starting CARMA background processes...
web-ui is up-to-date
Starting CARMA Platform foreground processes...
Starting lidar_lite_trailer_angle_driver
Starting ssc_controller_driver
Starting roscore
Starting velodyne-lidar-driver
Starting platform
Starting carma-cohda-dsrc-driver
Starting novatel-gnss-imu-driver
```
### Steps to reproduce the behavior

Ensure NAS logging is setup then run by mounting the drive and creating symlinks for the /opt/carma/logs and /opt/carma/.ros folders. ```carma start all```
### Related work

Link to related tickets or prior related work here.

TonysCousin, 2020-07-29:
@msmcconnell did you happen to check this on any other trucks?  Not that you were expected to.

msmcconnell, 2020-07-29:
@TonysCousin not yet. Probably won't get to it today

STOL-CARMA, 2020-09-01:
Added interface for NAS to the Switch network and this seems to have resolved the issue. Tested by running carma and confirming that it was logging to the NAS.

closed 2020-09-01 by STOL-CARMA