usdot-fhwa-stol/carma-platform #788: WindowsSharedFolder (NAS mount) does not have writeable permissions in White and Blue trucks

msmcconnell, 2020-07-30:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
The white and blue trucks seem to be able to communicate with the NAS without issues. However, carma cannot write to the logs or .ros folders. The following errors appear on all containers
```
avt-vimba-camera-driver            | No handlers could be found for logger "roslaunch"
avt-vimba-camera-driver            | [hardware_interface/driver_shutdown_dev_76_3533712415334403198-1] ERROR: unable to configure logging [unable to create directory for log file [/home/carma/.ros/log/cf3a3dc6-d292-11ea-919b-78d004284335].
avt-vimba-camera-driver            | Please check permissions.]
lidar_lite_trailer_angle_driver    | INFO: cannot create a symlink to latest log directory: [Errno 95] Operation not supported
avt-vimba-camera-driver            | [hardware_interface/camera/image_proc-2] ERROR: unable to configure logging [unable to create directory for log file [/home/carma/.ros/log/cf3a3dc6-d292-11ea-919b-78d004284335].
avt-vimba-camera-driver            | Please check permissions.]
```
I tried chmod to change the permissions but it didn't seem to stick even though no errors were thrown. 

The workaround for this is to remove the symlinks for /opt/carma/logs and /opt/carma/.ros and use real folders on the spectra instead. 
### Expected behavior
CARMA can log to the NAS
### Software version this applies to

develop after 3.3.0
### Actual behavior
See above
### Steps to reproduce the behavior
Run carma start all after ensuring symlinks exist to the WindowsSharedFolder for logs and .ros

### Related work

Link to related tickets or prior related work here.

msmcconnell, 2020-07-30:
Actually I might have misdiagnosed this. Notes incoming

STOL-CARMA, 2020-09-01:
updated /etc/rc.local mount command with forcegid and forceuid options, resolving permissions issue. Going to add the change to carma-platform/engineering_tools/rc.local and close the issue.

closed 2020-09-01 by STOL-CARMA