usdot-fhwa-stol/carma-platform #795: SSC license files on trucks are tied to specific binary instead of software version

msmcconnell, 2020-07-31:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
Testing between the Blue and White trucks has shown that the license files provided by AutonomouStuff for the SSC are tied to the specific SSC software binary installed on the vehicle and not to the SSC version as has been the case with the lexus. This means the current configuration management method for the SSC binaries will not work with the trucks. This leaves us with two options.
1. Get a license file that will validate with SSC version 1.1.1 for ssc_pm_cascadia package regardless of truck (not clear if this will cause any loss of performance in truck SSC) AStuff previously claimed the SSC could be used on any truck,

2. Create separate carma-config for each truck to launch a different SSC image and track the binaries separately. 
### Expected behavior
Same SSC binaries can be used on all trucks
### Software version this applies to

### Actual behavior
If you use the binaries from a different truck you get the following error from the SSC
```
Product license verification failed. Status:-1; No license for product
```
### Steps to reproduce the behavior

Run carma on the blue truck. 

### Related work

Link to related tickets or prior related work here.

msmcconnell, 2020-07-31:
Issue currently does not appear to be occurring on Silver truck so it might be isolated to blue.

msmcconnell, 2020-08-03:
AStuff ticket number 3218

msmcconnell, 2020-08-03:
Per AStuff support

"
ros-kinetic-ssc-pm-cascadia-dd13-228 - silver and white trucks
ros-kinetic-ssc-pm-cascadia-dd13-210 - red truck

ros-kinetic-ssc-pm-cascadia-dd13-180 - blue truck

The license on the blue truck is for ros-$ROS_DISTRO-ssc-pm-cascadia-dd13-180 so it will not work if the package from the white and silver trucks is installed.

"

maefromm, 2020-10-14:
@msmcconnell @jtbaird please update this issue with the latest updates you've received thus far.

jtbaird, 2020-10-14:
The red truck ssc binaries aren't compatible with pacmod_msgs 3.0.1 as described in the astuff dependency documentation, and any pacmod_msgs release prior to the creation date of the ssc binaries is using ros2. We either need a new binary that is confirmed to be compatible with a specific pacmod_msgs version, or for astuff to provide us with a compatible pacmod_msgs version for whatever is on the red truck. At this time it doesn't seem like any of the pacmod_msgs releases will work with the binaries unles we are willing to use ros2.

jtbaird, 2020-11-04:
Red truck pacmod_msgs dependency has been tested by Samir at Aberdeen and pacmod_msgs issue seems to be resolved. Closing issue.

closed 2020-11-04 by jtbaird