usdot-fhwa-stol/carma-platform #799: NAS data storage folder not available on Black Pacifica due to hardware issue.

snallamothu, 2020-08-04:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
The NAS data storage server on the Black Pacifica doesn't recognize the hard drives any more, it reports that drives are in "unconfigured bad" state at startup.  So, data cannot be written to the shared folder.
### Expected behavior
CARMA can log to the NAS.

### Software version this applies to
v 3.3.0 but this a hardware issue.

### Actual behavior
See description

### Steps to reproduce the behavior

1. Reboot server
2. Press delete when prompted to enter the setup utility
3. Browse to Advanced tab and click on hardware drive status.

### Related work
N/A

TonysCousin, 2020-08-04:
Marking this severity 3, as Carma can be fully functional with the NAS detached.  We are simply limited in the amount of testing that can be done between log deletions.

jtbaird, 2020-09-02:
RAID0 groups hard drives into virtual drives, and writes data across all of them simultaneously. Because of this, if one drive gets removed, the virtual drive is now broken, and all hard drives associated with it will show up as "unconfigured bad." The solution is to remove all the hard drives, put them in the IS300 server, go to the boot menu > MegaRIAD menu > drive management and then rebuild the drives.

In the future this can be prevented by removing/installing all of a virtual drive's hard drives simultaneously.

closed 2020-09-02 by jtbaird