usdot-fhwa-stol/carma-platform #805: autoware.ai install.sh script failing during automated builds

jtbaird, 2020-08-13:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
Docker Hub builds for autoware.ai are failing when install.sh tries to pull down lidar_localizer in step 4/18. See logs [here](https://hub.docker.com/repository/registry-1.docker.io/usdotfhwastoldev/autoware.ai/builds/52e31a48-5541-47b4-bdfa-9de50e021aa4).

### Expected behavior
autoware.ai dev image builds successfully.
### Software version this applies to

3.3.0
### Actual behavior
Build fails with following error:
[image]

### Steps to reproduce the behavior

run an automated build

### Related work

Link to related tickets or prior related work here.

TonysCousin, 2020-08-13:
Marking this as severity 2 because it prevents the software from being built, but there is a workaround whereby a developer can build the system directly from source, bypassing the docker image for Autoware (not desirable).

msmcconnell, 2021-11-08:
This has not been an issue for some time. Will close

closed 2021-11-08 by msmcconnell