usdot-fhwa-stol/carma-platform #806: Integration/Routing Branch - Route State and Merge Issues

maefromm, 2020-08-13:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
During the integration testing on #776, found the following issues: 
1. The route_generator package was added back incorrectly by the PR below. Only the route package should exist in the integration/routing branch. Please remove the route_generator package, files, and references. 
https://github.com/usdot-fhwa-stol/carma-platform/pull/753/files?file-filters%5B%5D=.h&file-filters%5B%5D=.txt&hide-deleted-files=true

2. After reviewing the latest Route detailed design document, found that the route_failed event has been renamed to route_gen-failed, and "route_invalidated" event was never implemented. Please add the logic for route_invalidated and update the design document. 

3. During integration testing, found that the route_state is not publishing at all. Please troubleshoot and fix the issue. 

### Expected behavior
1. route_generator package should not exist
2. route_invalidated logic is implemented and working, and the documentation for the route event diagram has been updated to reflect the latest. 
3. route_state should always be publishing. 

**NOTE: All of these fixes should be done first under integration/routing branch, and then merged into the integration/ui branch.** 

### Software version this applies to

integration/ui

### Actual behavior
Explained above. 

### Steps to reproduce the behavior
1. Build integration branch for carma-platform, carma-msgs, carma-web-ui. 
2. Run carma platform
3. Run carma-web-ui
4. On the UI, select and start the route. 
5. Confirm the route state and route events are working as expected. 
 
### Related work
#776,
Link to related tickets or prior related work here.

msmcconnell, 2020-08-14:
Adding onto this. There used to be a resource/ folder in the route package that contained the route needed for unit testing. It appears to be missing now.

dan-du-car, 2020-08-18:
To fix the route_state not publishing any data, upload below .osm file to /opt/carma/maps folder and create a symbolic link named  "vector_map.osm" point to this osm file. File zip is attached below, you will need to unzip:

[AOI_1_TFHRC_faster_pretty.zip](https://github.com/usdot-fhwa-stol/carma-platform/files/5093003/AOI_1_TFHRC_faster_pretty.zip)

dan-du-car, 2020-09-12:
@maefromm 
Please close this issue as the required changes were verified and merged.

msmcconnell, 2021-03-01:
Resolved by above PRs

closed 2021-03-01 by msmcconnell