usdot-fhwa-stol/carma-platform #810: Building with Colcon

jsun6065, 2020-08-14:
### Types of Issue

- [ ] Anomaly report (something appears to not work correctly)
- [X] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
Packages and docker images should be able to be built with colcon. 
### Expected behavior
Packages are built with colcon, the docker images are built with colcon and they are recognized.
### Software version this applies to

3.3.0
### Actual behavior
The packages and docker images both build with colcon.
### Steps to reproduce the behavior
Pull down the feature/building_with_colcon branch and run the build image script to verify the docker image build. In VSCode, build any package in the root of your workspace with "colcon build --packages-up-to <package name>".

msmcconnell, 2020-09-14:
All the currently used drivers have been merged, but platform, messenger, simulation, and CI are still needed. This can't be done until the message node is removed from the carma-messenger system.

msmcconnell, 2021-01-11:
All repos now use colcon. Closing.

closed 2021-01-11 by msmcconnell