usdot-fhwa-stol/carma-platform #834: Autoware Plugin not generating trajectory

saina-ramyar, 2020-08-31:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
During integration test it was noticed that the autoware plugin with external object interface does not generate trajectories and therefore the vehicle is not moving.
### Expected behavior
The autoware plugin should generate valid trajectories.
### Software version this applies to

3.3.0
### Actual behavior
During integration test it was noticed that the autoware plugin with external object interface does not generate trajectories and therefore the vehicle is not moving.
The autoware plugin without object avoidance logic is able to generate trajectories.
### Steps to reproduce the behavior

1. Do this
1. Then do this...

### Related work
#827 
Link to related tickets or prior related work here.

msmcconnell, 2020-09-03:
@TonysCousin You can probably downgrade this issue, now that the culprit PR has been reverted.

TonysCousin, 2020-09-11:
Changing to sev 2 since a workaround has been put into place, but the workaround removes desirable functionality, so this still needs to be resolved fairly soon.

msmcconnell, 2021-03-01:
Resolved by above PR

closed 2021-03-01 by msmcconnell