usdot-fhwa-stol/carma-platform #842: Unstable localization due to PR 816

msmcconnell, 2020-09-01:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
Testing on the develop branch on August 31st and Sep 1st 2020, showed unstable localization. Investigation of the issue tracked the problem to the changes merged by https://github.com/usdot-fhwa-stol/carma-platform/pull/816, which also had a small bugfix PR https://github.com/usdot-fhwa-stol/carma-platform/pull/831 . It is not immediately clear how this change is impacting localization performance as in its default operational mode the localization_manager node it added functions as a pass through. However, it was demonstrated that using the following commands to functionally revert that PR resulted in a return to acceptable performance.
```
# from carma-platform/ directory
git checkout c20dee3 gnss_ndt_selector carmajava/launch/localization.launch
rm -R localization_manager 
```
### Expected behavior
Localization performs stably
### Software version this applies to

develop after PR #816
### Actual behavior
Localization can be initialized but does not track stably when the vehicle is in motion
### Steps to reproduce the behavior

Run carma
localize
drive the vehicle

### Related work

Link to related tickets or prior related work here.

msmcconnell, 2020-09-03:
@TonysCousin You can probably downgrade this issue, now that the culprit PR has been reverted.

TonysCousin, 2020-09-11:
Downgraded to sev 2 since we have a workaround in place.  Unfortunately, the workaround is to disable other desirable functionality, so this is still a big deal.  But we can get on with our immediate release plans without fixing.

MishkaMN, 2021-01-12:
Resolved by PR https://github.com/usdot-fhwa-stol/carma-platform/pull/980

closed 2021-01-12 by MishkaMN