usdot-fhwa-stol/carma-platform #862: Lidar/Lidar calibration in Silver truck is off

msmcconnell, 2020-09-18:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

While running CARMA Platform in the silver truck it was observed that the lidar scans are clearly not aligned as features we showing up as duplicated. 

### Carma version where this issue was discovered

CARMA Platform develop after 3.3.0
### Expected behavior
Both lidar scans should overlap so they appear as a unified view of the world.
### Actual behavior

Lidar scans clearly show duplicated features
[image]

### Steps to reproduce the actual behavior
carma start all
Launch rviz
localize
Look at scans in rviz

### Related work

TonysCousin, 2020-09-18:
Marking sev 2 as it prevents localization (a critical function), but the workaround is to use another truck.

TonysCousin, 2020-09-30:
Cam Brown of AStuff recalibrated it on 9/30/20.

TonysCousin, 2020-10-01:
Closing now that the PR is merged.

closed 2020-10-01 by TonysCousin