usdot-fhwa-stol/carma-platform #864: Remove java version message node launch from message.launch file

dan-du-car, 2020-09-22:
### Types of Issue

- [ ] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

### Carma version where this issue was discovered

### Expected behavior

### Actual behavior

### Steps to reproduce the actual behavior

1. Do this
1. Then do this...

### Related work

msmcconnell, 2020-09-23:
Java message node is not yet deprecated as BSM/Map/SPaT all still require it. However, it is worth investigating if the message node and cpp_message node are truly interoperable.

TonysCousin, 2020-09-28:
Marking sev 1 since it seems that no communications are possible at this time, and Carma is all about communicating.

closed 2020-10-06 by dan-du-car