usdot-fhwa-stol/carma-platform #870: Silver Lexus: left camera holder broken

MishkaMN, 2020-09-24:
### Types of Issue

- [X] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

When I was integration testing in the silver lexus, I accidentally broke the clipper that holds the left camera while trying to raise the monitor higher. The camera itself is fine, and CARMA is operational without it. Currently it can sit by itself on the clipper although it is broken, but a sudden jerk will make it fall off. Either it needs to be glued back or replace the holder. And the camera may need to be calibrated again.

### Carma version where this issue was discovered

not a software issue, but CARMA 3.3.0+
### Expected behavior
Left camera is placed back firmly and recalibrated on Silver Lexus.

### Actual behavior

The camera can fall off while driving. 

### Steps to reproduce the actual behavior

See above.

### Related work

maefromm, 2020-09-28:
Opened AStuff ticket. "Your request (3431) has been received and is being reviewed by our support staff."

MishkaMN, 2020-11-05:
This particular issue was resolved.

closed 2020-11-05 by MishkaMN