usdot-fhwa-stol/carma-platform #879: Tahoe - OBU to DSRC to CARMA Truck Mobility Operation Message Forwarding Issue

maefromm, 2020-09-29:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

During the Truck Inspection Plugin integration testing by Dan using the Silver Truck and Tahoe, the message wasn't fully forwarded by the Tahoe's DSRC Driver to the Message Node. Dan/Abey/Mike arrived to the conclusion that the issue may lie in here https://github.com/usdot-fhwa-stol/carma-cohda-dsrc-driver/blob/develop/dsrc_driver/src/dsrc_client.cpp

On 9/25/20: Anjan and I confirmed that the entirety of the payload of the messages is being sent and received by the OBUs. Using the ASN schema provided to us by Mike, we were able to successfully decode the entire message using asn1playground. The schema and decoded message are attached. I think that we can confidently say that the OBUs and the ASN schema are working as intended. This seems like an issue on the decoding side of the CARMA code. 

[image]

### Carma version where this issue was discovered

develop branch

### Actual behavior

1. Start carma on both truck and tahoe. 
2. From Silver Truck, send the truck safety info.
rostopic pub -r 10 /message/outgoing_plain_mobility_operation_message  cav_msgs/MobilityOperation "header: {sender_id:SDOT-45100', recipient_id: 'USDOT-45095', sender_bsm_id: '10ABCDEF', plan_id: '11111111-2222-3333-AAAA-111111111111', timestamp: 9223372036854775807}
strategy: "TruckInspection"
strategy_params: 'vin_number:1FUJGHDV0CLBP8896,carrier_name:FMCSA Tech,date_of_last_state_inspection:2020.01.20,date_of_last_ads_calibration:2020.02.20,license_plate:DOT-10003,state_short_name:US,weight:65000,carrier_id:1,permit_required:Yes,iss_score:75,pre_trip_ads_health_check:red,ads_health_status:1,ads_auto_status:engaged,ads_software_version:CARMA Platform v3.3.0'"

3. From Tahoe, listen to the incoming binary message. Currently, it is showing "Type: Unknown"

### Expected behavior
If Tahoe OBU receives the full packet from the Silver Truck OBU, then DSRC driver and Message node should be able to decode to decode the full message. 

Same steps above. 
3. From Tahoe, listen to the incoming binary message. It should NOT be showing "Type: Unknown"
4. From Tahoe, the Message Node should be able to decode the full message 
5. From Tahoe, UI should be able to display the Truck Safety Info fully. 

### Steps to reproduce the actual behavior

See above. 

### Related work

maefromm, 2020-09-29:
[d600a0c6e7124637a2631c3212119fde.hex.uper.txt](https://github.com/usdot-fhwa-stol/carma-platform/files/5301015/d600a0c6e7124637a2631c3212119fde.hex.uper.txt)

dan-du-car, 2020-09-29:
From Tahoe. The received data size from dsrc driver receives data size around 417 bytes, however it publishes only 4 bytes to /comms/inbound_binary_msg.
[image]

TonysCousin, 2020-09-29:
Assigning sev 1 because it appears the DSRC driver is unable to forward incoming MobilityOperation messages, which are central to cooperative behavior of any kind.

maefromm, 2020-09-30:
Temporary fixes has been applied by @dan-du-car to carma-platform and carma-messenger under this sprint. 
Plan is to revert these changes once the dsrc driver issue is fixed. 

https://github.com/usdot-fhwa-stol/carma-platform/pull/882
https://github.com/usdot-fhwa-stol/carma-messenger/pull/48

maefromm, 2020-10-05:
As discussed with @dan-du-car @adev4a @kjrush, we will also update the header to be properly filled out by carma-messenger whenever there is a mobility request or mobility operation sent out. At a minimum, I think sender ID, recipient ID and timestamp. 

Original design by original developer was leaving the headers blank, but discussed with Kyle that some basic header information is required. 
[image]

closed 2020-12-08 by dan-du-car