usdot-fhwa-stol/carma-platform #901: Red truck binaries do not match pacmod version in carma-base

jtbaird, 2020-10-09:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

After updating the truck binaries per issue #795, we encountered an error (see Actual Behavior below) stating that the veh_interface binary was expecting a different version of the pacmod_msgs package than what is currently being installed in carma-base.

### Carma version where this issue was discovered

CARMASystem_3.3.0
### Expected behavior
SSC engages successfully and is compatible with the pacmod.

### Actual behavior

When carma is run on the red truck, the following error is thrown by the SSC:

`ssc_controller_driver              | [ERROR] [1602191235.802347293]: Client [/hardware_interface/veh_interface] wants topic /hardware_interface/pacmod/parsed_tx/shift_rpt to have datatype/md5sum [pacmod_msgs/SystemRptInt/9fc8568a79d68cb7474969c688231163], but our version has [pacmod_msgs/SystemRptInt/da718c3df94de6df3ce729dd13078d8a]. Dropping connection.`
### Steps to reproduce the actual behavior

1. Clone a specific version of pacmod_msgs from source in carma-ssc-interface-wrapper/docker/checkout.bash
2. Build a new ssc image
3. Test
4. If the error persists, roll back to an earlier version of pacmod_msgs and try again.

### Related work

Should be exactly the same as #744

TonysCousin, 2020-10-23:
Marking as sev 3 since it only affects one vehicle in the fleet.

TonysCousin, 2020-11-02:
Red truck successfully moved under automated control with code from the develop branch on 10/30/20, demonstrating that this issue has been resolved.

closed 2020-11-02 by TonysCousin