usdot-fhwa-stol/carma-platform #908: health_monitor publishes system alerts continuously

msmcconnell, 2020-10-14:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

System alert publication from health_monitor is periodic (10Hz), but should be event driven.
The health_monitor publishes system alerts in a periodic fashion. This causes issues with the UI which is not expecting a continuous stream of error messages. 

Instead of publishing the same message at 10Hz the health_monitor should publish alerts only once on a latched publisher to guarantee all nodes receive the most recent alert. 

### Carma version where this issue was discovered

develop after 3.3.0
### Expected behavior
System alerts are latched and published once
### Actual behavior

Health monitor continues to publish the same system alert until a status change occurs. 
### Steps to reproduce the actual behavior

Monitor system_alert topic at any point during system operation. 

### Related work

This issue was created by splitting https://github.com/usdot-fhwa-stol/carma-platform/issues/865

TonysCousin, 2020-11-09:
I think it's a little more complicated than the description indicates.  If a sensor goes offline, then, indeed, we only want a single message published to indicate that.  But if it comes back to life and the system starts using it again, we want to track that state, so if it goes offline a second time, we should see a second message about that new event.

I'll acknowledge that, for now, the system probably is not sophisticated enough to handle a sensor failing then coming back online, even if health monitor reported it so.  Therefore, until such resiliency is built in, doing the latch as described should be okay.

msmcconnell, 2021-03-01:
Resolved by above PR.

closed 2021-03-01 by msmcconnell