usdot-fhwa-stol/carma-platform #916: CARMAWebUi - Fix the Widget View and Side Bars Issues

maefromm, 2020-10-21:
### Types of Issue

- [ ] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

For the ACE Demo, 
1) Fix at a minimum the widget view to showcase the crusing plugin widget with the speedometer that's currently missing due to CARMA Guidance using Autoware Plugin. 

2) Remove the obsolete sidebar icons that are no longer needed for the ACE Demo such as the lat and lon plugin indicator, the ACC indicator, and others. 

### Carma version where this issue was discovered

Develop Branch. 

### Expected behavior
Should display Widget View with the speedometer and other items that were under CARMA2 Cruising plugin has displayed before that are still applicable. 

### Actual behavior

Nothing is shown on Widget View and some of the side bar icons are obsolete and not populated. 

### Steps to reproduce the actual behavior

Run CARMA UI
Engage CARMA
Review the Side Bars to ensure it's working order
Review the Widget View to ensure it's working order. 

### Related work

For ACE Demo

TonysCousin, 2020-10-23:
Missing widget view drives this to be sev 3.  Extraneous sidebar icons are sev 4 things.

dan-du-car, 2020-11-02:
Could you specify which status icons exactly are not needed?

msmcconnell, 2021-03-01:
Resolved by above PRs

closed 2021-03-01 by msmcconnell