usdot-fhwa-stol/carma-platform #917: white truck new yaml file with update Yaw value

itabrl, 2020-10-21:
### Types of Issue

- [ ] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

white truck new yaml file with update Yaw value needs to be copy over in the vehicle calibration repository. 

### Related work

white truck testing

TonysCousin, 2020-11-02:
@msmcconnell I see this as a routine maintenance issue, as sensors are going to go out of calibration occasionally.  I could see an argument for making it either an anomaly or an "other".  In retrospect, it does not seem that enhancement is appropriate, however.  I would argue against anomaly, as that implies a factory fault.

TonysCousin, 2020-11-02:
Tests at ATEF on 10/28/20 - 10/30/20 demonstrated that the new calibration value is appropriate and the lidar is functioning well.

closed 2020-11-02 by TonysCousin