usdot-fhwa-stol/carma-platform #918: Fix CARMAWebUi - Listen to System Alert Shutdown instead of System Alert

maefromm, 2020-10-21:
### Types of Issue

- [x ] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

Per @msmcconnell, the latest logic or thought process with health monitor for CARMA3 is different from CARMA2. Currently from CARMA2, the UI would listen to FATAL alerts since the assumption is when Guidance sends the fatal alerts or any node, the full system "should" shut down. 

With the new health monitor that is no longer the case. The health monitor should be the one issuing the shutdown for the system when the node issues a fatal and shuts only itself down. If health monitor determines it's a critical node or component, then it will issue the shutdown. 

Per discussion, the changes to health monitor will forward the FATAL description when SHUTDOWN occurs so that the UI will see the relevant error associated with the shutdown. This is done separate from this issue. 

The change on UI is to ensure it does not shutdown the UI when FATAL is received. Instead:
- When FATAL occurs, a new "critical error" shall be displayed. 
- When SHUTDOWN occurs, a popup will be shown that system has shut down. 

### Carma version where this issue was discovered

DEVELOP 

### Expected behavior
See above. 

### Actual behavior

See above.

### Steps to reproduce the actual behavior

Run the latest Health Monitor
Run the latest UI from this issue

### Related work

https://usdot-carma.atlassian.net/browse/CAR-2035

TonysCousin, 2020-10-23:
@maefromm: Health monitor does not need to "forward" a fatal message for UI to receive it.  Anything published on system_alert topic by any node is already getting to the UI.  Is there a need for the description field to be handled differently?

msmcconnell, 2020-10-27:
As discussed offline, health_monitor should have arbitration responsibility for system shutdown and there fore the UI should only display a poppop on shutdown messages as described by this issue. Closing as this was resolved by the above PR

closed 2020-10-27 by msmcconnell