usdot-fhwa-stol/carma-platform #937: lightbar is not automatically controlled by carma (first noticed on trucks)

MishkaMN, 2020-10-29:
### Types of Issue

- [X] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

It was brought to our attention that the lightbar does not correctly reflect the automation states of the carma on trucks.
Involved components are lightbar_driver and lightbar_manager.
Lightbar_manager was never tested on passenger cars, but only was on mock drivers, so potential error could be there.

This does not affect functional operation of the carma, but introduces a potential safety concern as it removes visual indication of the car's automation states.

### Carma version where this issue was discovered

carma 3.3.0+ develop
### Expected behavior
Essential states:
When the car starts, or carma inactive, all lights should be off. 
When the carma is engaged, the green should on (solid).
When the carma is disengaged, but active, the green flash should on.

In other words, expected state transition:
[image]
Expected different patterns:
[image]

### Actual behavior

Lights are on, but it doesn't reflect the various states of the automation.

### Steps to reproduce the actual behavior

Pull every develop branches and run carma start all.

### Related work

TonysCousin, 2020-10-29:
In particular, this issue was brought up by Brian Reed at ATEF, who is concerned about its implications for safety on that track.

MishkaMN, 2020-11-09:
The issue is resolved by above PRs.
Any further issue is most likely related to lightbar wiring mismatch or reprogrammability

closed 2020-11-09 by MishkaMN