usdot-fhwa-stol/carma-platform #948: Version ID not displaying on Wanderer UI

TonysCousin, 2020-11-09:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

When starting carma and displaying the UI, the info bar at the bottom of the main page, where version ID is supposed to appear, simply says "NULL".

### Carma version where this issue was discovered

Wanderer

### Expected behavior
Version ID displays on UI

### Actual behavior

UI displays "NULL" instead.

### Steps to reproduce the actual behavior

1. Start Carma, look at UI main screen.

### Related work

TonysCousin, 2020-11-09:
This issue is confirmed on both white and blue trucks running latest version of Wanderer candidate.

maefromm, 2020-12-02:
The version id is not showing on Docker Hub builds only. Seems the build.sh -d works just fine. Per JB, "So the docker hub builds currently just use a docker build command to create the images, so they don't have all of the metadata that is added via our build script. I've created a fix for this but I talked with John and we decided it would be best not to add the changes until after the release." So the command below will not work at this time that web-ui uses on both wanderer and basic travel ui

dev@dev-Nuvo-6108GC:~$ docker container inspect -f '{{ index .Config.Labels "org.label-schema.version"}}' carma-config
develop-development

So that's why version ID in the web-ui is showing NULL at this time

maefromm, 2020-12-14:
The metadata is also missing on carma-messenger-ui, and fix applied above.

msmcconnell, 2021-03-01:
Resolved by above PRs

closed 2021-03-01 by msmcconnell