usdot-fhwa-stol/carma-platform #969: SSC Controller frequently timeouts with a beeping noise

maefromm, 2020-11-16:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

Noticed SSC Controller times out often and produces a beeping noise every time the vehicle is in idle and manual mode. 
-	Silver Lexus – Happened 11/11multiple times. 
-	Blue Lexus – Per Mike he saw this happened on 11/10. 
-	White Truck – 11/11 per Sudhakar: “when this happened on the white truck yesterday morning, we had to restart the truck because the beeping noise wouldn't go away even after resetting using the e-stop.”

### Carma version where this issue was discovered

### Expected behavior
When vehicle is manual control and idle for a long time, SSC should not timeout. 

### Actual behavior

As described above. 

### Steps to reproduce the actual behavior

1. Start CARMA. 
2. Be on idle for a while and wait for the SSC to time out.  

### Related work

Opened a ticket with AStuff Ticket 3569

msmcconnell, 2021-03-22:
Haven't seen issues on silver lexus during last few days of testing.

msmcconnell, 2022-01-12:
@maefromm Do you know if this is still a thing?