usdot-fhwa-stol/carma-platform #993: Truck DSRC communication issue during ACE Demo Verification Testing

maefromm, 2020-12-07:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

### Carma version where this issue was discovered

Wanderer Release Candidate 

### Expected behavior
Final approach to be discussed, see comments below. 
In the interim, it was discussed that the antenna that broadcasts on 172 needs to move to the left side of the truck. 

Please do not close this issue until final resolution has been agreed upon and tested. 

### Actual behavior

**Notes from Verification Testing on 11/30 to 12/3 at ATEF:** 
After lunch, we turned on the electronics in the Tahoe and setup carma-messenger in the parking lot. We verified that the message requests could be sent and received on both vehicles and the reported data and statuses were correct. This worked without issue, so we moved onto the track. We did 4 runs of the full demo scenario (starting with lidar->gps) using the White truck and 3 runs with the Red truck. To stress test the DSRC communications, I made a point of requesting the log multiple times during each run. In all but one run, I was able to get the log at least once. I was even able to request it 5 times in one run on the Red Truck. However, the DSRC comms were somewhat intermittent with at least 1 request failing in 4 of the 8 runs. I believe this is largely due to signal blockage and hardware failures as described in more detail farther down. 

**White Truck**
- Strongly suspect that the left antenna is not broadcasting DSRC. The signal was far more reliable on the right hand side and almost non-existent if the right antenna was not visible. 
- No DSRC reception behind trailer. Trailer antennas must not be connected. 

**Red Truck**
- No DSRC reception behind trailer. Trailer antennas must not be connected.

### Steps to reproduce the actual behavior

See actual behavior. 

### Related work

maefromm, 2020-12-07:
Friday, Dec 4th conversation. 
[EML] The two antennas are broadcasting/receiving on different channels and only broadcast/receive from the antenna that matches the channel that you are using.  So if all comms are going out on ch 172 (or some other) then only 1 antenna is used and this behavior sounds normal.
Regarding the trailer OBU working as ‘slave’ to the cab OBU, do you mean CARMA sends messages to the cab OBU and then they are forwarded to the trailer OBU?  I don’t think Cohda’s fw can do that as a built in feature, so it might be easier for CARMA to send it to both if it’s not already.
[John] thought was to use only the cab OBU and connect it to all 4 antennas.  Maybe that’s not possible? We wanted to avoid having Carma send outbound messages to both OBUs, as that would double the airwave bandwidth use and also require a receiving unit to process duplicate messages, which might trip it up in some cases.  
[EML} We talked a while back about investigating that.  It’s possible to split/combine the signals between two different antennas, but it requires extra hardware and we need to be careful about things like power levels and cable lengths.  If we want to take the time to look at that, we might also consider splitting the signals between both sides of the cab and we might avoid having antennas on the trailers.

TonysCousin, 2020-12-07:
Marking this as sev 2 since positioning the vehicles in a different orientation provides a good workaround for getting radio signal visibility.

msmcconnell, 2020-12-17:
It seems that it is design intent for only one antenna to be broadcasting DSRC due to the way OBUs work. We have ensured that the left antenna is broadcasting on the white truck and blue truck seems to perform very reliably as well. So as far as the ACE demo is concerned this issue is resolved, however, a longer term solution is still needed.

msmcconnell, 2021-03-01:
Closing the anomoly issue per the above commend and opening an enhancement issue to address multi-antenna broadcast

closed 2021-03-01 by msmcconnell