usdot-fhwa-stol/carma-platform #998: Red Truck NDT localization not reliable on ATEF track

msmcconnell, 2020-12-07:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

The NDT localization on the Red truck is not very stable on the ATEF track. An abort is always needed at the ends of the runway and sometimes at the start if the initial motion in jerky enough. This is in contrast to the white truck which is able to consistently drive the demo route under NDT localization. 
We have recently recalibrated the lidar->lidar calibration, so an alternative cause is likely. 
### Carma version where this issue was discovered

release/Wanderer
### Expected behavior
All trucks perform similarly and NDT can be used for follow the ACE demo route without an override. 

The current work around for this issue is to run the demo using GPS. 
### Actual behavior

Override always required at ends of runway. 
### Steps to reproduce the actual behavior

Try driving the ACE demo route with the Red Truck

### Related work

TonysCousin, 2020-12-07:
Marking as sev 2 because localization is a fundamental capability.  Even though it is only on one truck, it takes 1 of 3 out of action.  It appears to be a vehicle-specific problem, since white truck works well, but until that is more confidently proven, I'm hesitant to attach the Vehicle label.

msmcconnell, 2021-03-15:
Need to test but the new GPS with NDT initialization mode might make this issue moot.

msmcconnell, 2022-02-04:
Usage of the GPS with NDT initialization provided sufficient accuracy and reliability to complete the verification testing of Battista. So will close

closed 2022-02-04 by msmcconnell