usdot-fhwa-stol/carma-platform #1002: Rosbridge failing due to dependency requirement

MishkaMN, 2020-12-09:
### Types of Issue

- [X] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

rosbridge is failing because it cannot find builtin package as rosbridge was updated to require that (when we updated VTK).
### Carma version where this issue was discovered

release/Wanderer branch
3.3.0+
### Expected behavior

### Actual behavior

It crashes when it starts
### Steps to reproduce the actual behavior

carma start all while on release/Wanderer branches

### Related work

created a ticket on rosbridge https://github.com/RobotWebTools/rosbridge_suite/issues/544

TonysCousin, 2020-12-09:
Marking sev 2, as the UI is a critical functionality.  But we can work around it (ugly) by using command line to invoke Carma and watch whatever variables are desired.  It is not what the customer will be happy with, but it will allow testing to continue until a fix is identified.

msmcconnell, 2020-12-10:
Resolved by PR #1001

closed 2020-12-10 by msmcconnell