usdot-fhwa-stol/carma-platform #1028: Trajectory_executor issues error "Ran out of trajectory" at route end

adev4a, 2021-01-11:
### Types of Issue

- [x ] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

On implementing the end of route behavior, which is a slow down and wait planned by the stop_and_wait_plugin, the guidance state is switched to Inactive and consequently DRIVERS_READY. When the vehicle is manually moved out of the stop state, the error "Ran out of trajectory data to consume!" is published by the trajectory executor.

### Carma version where this issue was discovered

3.3.0

### Expected behavior
Carma should allow another route to be planned.
### Actual behavior

An error "Ran out of trahectory data to consume" is issued by the trajectory executor. It should be noted that the stop in this case 
is happening before the end of the known route. A stop within the of ~10meters has been defined as an acceptable range for route completion.
### Steps to reproduce the actual behavior

1. Run carma over a defined route. 
2. At the end of the route, after the vehicle comes to a complete stop and the UI issues the message "Route Completed", manually disengage carma, and drive further.

### Related work

https://usdot-carma.atlassian.net/browse/CAR-3827

kjrush, 2021-02-22:
Awaiting validation that this issue is still present in vanden-plas release candidate

msmcconnell, 2021-03-22:
Resolved by above PR

PR #1135 (2021-03-07): Fix state check in guidance state callback
Guidance state updates were being checked against == INACTIVE which may miss some cases in which the trajectory should not be consumed. This has been fixed to check against != ENGAGED. A redundant callback (unused) that contained this logic has been removed and the correct callback has been updated with this logic.

# PR Details
## Description

Implements a fix in guidance state processing logic inside of the trajectory executor. Previously trajectory processing was stopped only on transition to INACTIVE. Now it is stopped on transition to any state o [... cut]
closed 2021-03-22 by msmcconnell