usdot-fhwa-stol/carma-platform #1029: Circle CI is not failing the PR even after detecting unit test failures

MishkaMN, 2021-01-11:
### Types of Issue

- [X] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

- When scrolling  through the unit test workflow, the CircleCi is not failing the PRs with unit test failures. 
- Also we noticed some unit tests are not being detected at all (0% coverage) by Sonar. 

https://app.circleci.com/pipelines/github/usdot-fhwa-stol/carma-platform/3297/workflows/355d3b6c-64bd-49ab-9365-e9361e29f34e/jobs/3439

Examples are:
[image]
[image]
[image]

etc. The developer should find all those remaining and resolve it.

### Carma version where this issue was discovered

carma-platform develop 3.4.0 +
### Expected behavior
there should be no unit test failures.
### Actual behavior

unit tests fail not CircleCi doesn't detect it.
new unit tests sometimes are not detected by sonar.

### Steps to reproduce the actual behavior

Put up a PR and see CircleCI, Sonar reports.

### Related work

msmcconnell, 2022-09-29:
This is still occurring with ROS2 packages but now the output looks like this
]0;colcon test [34/110 done] [4 ongoing]Finished <<< carma_wm_ros2 [2.60s]	[ with test failures ]

msmcconnell, 2022-09-29:
@codygarver @SaikrishnaBairamoni 
Please add this onto your devops backlog