usdot-fhwa-stol/carma-platform #1036: Guidance tactical plugins cannot handle interchanging maneuver plans

adev4a, 2021-01-15:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

The plan delegator sends a complete list of requested maneuver plans to tactical planners.
Since each tactical plugin can only create a plan for a specific maneuver, this creates an issue when planning for multiple maneuvers.

A fix introduced by Issue 1023, provides a temporary fix that allows the stop and wait maneuver to planned by the stop_and_wait_plugin, but it only works for the end of a route.
This fix would not work when a stop and wait maneuver is requested in the middle of a route.

### Carma version where this issue was discovered

3.3.0

### Expected behavior
Changing maneuver plans can be made in strategic plugins and tactical plugins can generate trajectories for them corresponding to their actual order in the plan.
### Actual behavior

### Steps to reproduce the actual behavior

1. Edit route_following to have a stop_and_wait maneuver in the middle of the route.

### Related work

msmcconnell, 2021-03-01:
Related to https://github.com/usdot-fhwa-stol/carma-platform/issues/1106

JonSmet, 2021-04-30:
Closed by PR #1220 in carma-platform and PR #102 in carma-msgs.

closed 2021-04-30 by JonSmet