usdot-fhwa-stol/carma-platform #1057: Cannot restart same route after getting LEFT ROUTE event

msmcconnell, 2021-02-05:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

After getting a LEFT ROUTE event, I was unable to reselect the same route on the UI. I recieved the ALREADY_FOLLOWING_ROUTE error. 

### Carma version where this issue was discovered

release/vanden-plas
### Expected behavior
You can restart the same route after leaving it. 
### Actual behavior

See description
### Steps to reproduce the actual behavior

1. Select route and drive
2. leave route
3. Got LEFT ROUTE notification on UI and hit Restart
4. Selected route again from UI
5. Got ALREADY_FOLLOWING_ROUTE notification on UI and was not able to reselect the route.

### Related work

msmcconnell, 2021-02-16:
Resolved by above PRs

dan-du-car, 2021-03-08:
reopen due to another pending PR

dan-du-car, 2021-03-09:
Close per PR merged

closed 2021-03-09 by dan-du-car