usdot-fhwa-stol/carma-platform #1059: Silver and Blue lexus cannot make left hand turn at East Intersection at TFHRC

msmcconnell, 2021-02-05:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

Niether the silver or blue lexus can safely make the left hand turn at the easy intersection. The vehicles approach the turn at far too high of a speed. 
Observaton of the map, route, and trajectory position all show them to be reasonable. This implies the issue lies in the speed generation approach. 
### Carma version where this issue was discovered

release/vanden-plas
### Expected behavior
Vehicle can make all turns
### Actual behavior

See description
### Steps to reproduce the actual behavior

Drive vanden-plas test plan route through east intersection at TFHRC

### Related work

msmcconnell, 2021-03-01:
Resolved by above PR

lewisid, 2021-03-09:
During testing, the Blue Lexus failed to make the left-hand turn at the second stoplight/east intersection at TFHRC it would drive too fast when approaching the entrance and make a very wide turn (almost ending up on the sidewalk).

msmcconnell, 2021-03-10:
Per @lewisid comment, I believe this issue has reemerged as of vanden-plas commit 6f046fae17ed69ae5979d94c6f48d91fcd6c6cea though the cause could come from one of the other repos so the date on the commit would need to be compared.

msmcconnell, 2021-03-15:
After deploying the most recent vanden-plas build with default parameters this was not an issue. Will reclose this issue.

closed 2021-03-15 by msmcconnell