usdot-fhwa-stol/carma-platform #1087: motion_computation error

itabrl, 2021-02-10:
### Types of Issue

- [ ] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

Running carma on white pacifica shows bellow error :
> platform                   | 1612989767.781666840 | ERROR | Buffer::checkAndErrorDedicatedThreadPresent:195 | Do not call canTransform or lookupTransform with a timeout unless you are using another thread for populating data. Without a dedicated thread it will always timeout.  If you have a seperate thread servicing tf messages, call setUsingDedicatedThread(true) on your Buffer instance.
> platform                   | 1612989767.781808810 | ERROR | CARMANodeHandle::handleException:201 | Uncaught Exception in /environment/motion_computation exception: Look up transform timed out: Do not call canTransform or lookupTransform with a timeout unless you are using another thread for populating data. Without a dedicated thread it will always timeout.  If you have a seperate thread servicing tf messages, call setUsingDedicatedThread(true) on your Buffer instance.

### Carma version where this issue was discovered

develop 

### Steps to reproduce the actual behavior

run carma on develop 

### Related work

msmcconnell, 2021-02-11:
Deverity 1 for develop, but does not affect vanden-plas

msmcconnell, 2021-02-15:
Resolved by above PR

closed 2021-02-15 by msmcconnell