usdot-fhwa-stol/carma-platform #1093: Ford Fusion disengagement due to veh_controller error not reported on UI

JonSmet, 2021-02-11:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

On the Ford Fusion, there were two separate occurrences during Vanden Plas verification testing in which /hardware_interface/module_states issued a "Drive by Wire Gear Module Fault" warning, which causes the system to disengage and coast in neutral. There was no notification from the Web UI that manual control was required.

### Carma version where this issue was discovered

release/vanden-plas

### Expected behavior
It is expected that a notification be displayed on the Web UI to alert the driver that manual control is required when a "Drive by Wire Gear Module Fault" warning is issued, which caused the system to disengage and coast in neutral.

### Actual behavior

See above for actual behavior.

### Steps to reproduce the actual behavior

Behavior occurs intermittently on Ford Fusion (approximately 1 in 5 route attempts during Vanden Plas testing). 

### Related work

msmcconnell, 2021-03-15:
This was duplicated in #1141 which contained additional details now copied here
### Descriptive summary

Opened AStuff Ticket #3793 with the following issues description: 
Good morning, Cullen and Ben. 
I have received reports on SSC errors/faults on the Ford Fusion quite frequently. Could you help us address the SSC issues we’re having on Ford? 
**Yesterday from Sudhakar:** 
1.       We are getting a lot of SSC faults on Ford Fusion multiple times in a trip. We also noticed that there is SSC log message in the CARMA terminal that indicated that there is a new firmware. Can we do this update on the air or how should we do this? Attached is a screenshot provided in the foler below: “20210309_SSCFault_NewFirmwareNotification”
**Today from Misheel:** 
2.       At first I thought this was not related to ssc error, as I thought UI is reporting ssc failed when everything failed because of wmbroadcaster not being able to find map.But as it is happening pretty frequently, I thought it was best to have it checked by Astuff. See rosbag “fushion-ssc-err-at-start.active” and “ssc-error-snippet” text log file provided in the folder below.
3.        There is another type of ssc error I have a rosbag for (that happens in mid-run), but its bag is big, so it is taking time in uploading to drive. 
4.       There a silent ssc error as well that occurs: fushion-silent-ssc.bag.active
 
https://drive.google.com/drive/folders/1bon2VSe4EGHj33PpFgslt_ayPrdGuMLw?usp=sharing

### Carma version where this issue was discovered

Vanden Plas Candidate Release 

### Expected behavior
Should not recieve any frequent SSC faults.

msmcconnell, 2021-04-13:
According to AStuff, the DataSpeed controller license might have been causing issues. It has been updated now and this issue should be gone. We will keep this open for a bit longer and if it does not occur during testing over the next week or so, we will close it.