usdot-fhwa-stol/carma-platform #1117: roadway object plugin issue

itabrl, 2021-02-26:
### Types of Issue

- [`x`] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

with correct input to the package there is empty list as an output

### Carma version where this issue was discovered

develop
### Expected behavior
output with the location of the vehicle front of the carma car

### Actual behavior

empty list as an output

### Related work

msmcconnell, 2021-03-01:
Related to https://github.com/usdot-fhwa-stol/carma-platform/issues/1068

msmcconnell, 2021-03-15:
Resolved by PR #1145

closed 2021-03-15 by msmcconnell