usdot-fhwa-stol/carma-platform #1130: Fix CARMA Platform Required Plugins to fix Web UI display

maefromm, 2021-03-02:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

Fix the required_plugins configuration file to fix the UI display of required plugins to have Route Following, InLaneCruisingPlugin, StopandWaitPlugin, Pure Pursuit.

### Carma version where this issue was discovered

Vanden Plas Candidate

### Expected behavior
The picture below should show InLaneCruisingPlugin, StopandWaitPlugin as required as well. Only 4 plugins are showing currently on Vanden Plas release and all four are required. 
[image]

### Actual behavior

See above.

### Steps to reproduce the actual behavior

See above. 

### Related work

dan-du-car, 2021-03-03:
@maefromm There is no space between below plugin names: RouteFollowing, InLaneCruisingPlugin, StopandWaitPlugin. Do we need to update this name with spaces?

maefromm, 2021-03-11:
@dan-du-car can this now be closed based on your integreation testing in vehicle?

msmcconnell, 2021-03-22:
Resolved by above PR

closed 2021-03-22 by msmcconnell