usdot-fhwa-stol/carma-platform #1146: Unable to run CARMA platform

DokurOmkar, 2021-03-12:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

I have set up a native Ubuntu 16.04 machine with the following configuration (8 core processor, 24GB RAM, GeForce 1650 Graphics) and followed the confluence page [here](https://usdot-carma.atlassian.net/wiki/spaces/CRMPLT/pages/470286338/Setup+a+VM+Ware+Image+for+Development) to set up the username, password, etc.

[image]

And I followed the entire documentation to set up the CARMA platform and then I am seeing several warnings after running `carma start all`

The warnings are:

[image]

`WARNING: Error loading config file: /home/carma/.docker/config.json: open /home/carma/.docker/config.json: permission denied`

[image]

`roscore                       | WARNING: Could not change permissions for folder [/opt/carma/logs/8463e2a6-8364-11eb-aa37-5cf9dd6b91fb], make sure that the parent folder has correct permissions.`

[image]

`carma-mock-lightbar-driver    | 1615575336.685362033 | FATAL | main:188 | Error opening file: /opt/carma/data/mock_drivers/drivers.bag`

[image]

`platform                      | process[environment/lanelet2_map_visualization-19]: started with pid [215ERROR: cannot launch node of type [motion_computation/motion_computation_node]: can't locate node [motion_computation_node] in package [motion_computation]`

With this, I have opened the localhost/index.html. When I clicked on Start Platform button, I am seeing errors in the terminal which is below:

`
platform                      | 2021-03-12 18:56:00+0000 [-] Log opened.
platform                      | 2021-03-12 18:56:00+0000 [-] registered capabilities (classes):
platform                      | 2021-03-12 18:56:00+0000 [-]  - rosbridge_library.capabilities.call_service.CallService
platform                      | 2021-03-12 18:56:00+0000 [-]  - rosbridge_library.capabilities.advertise.Advertise
platform                      | 2021-03-12 18:56:00+0000 [-]  - rosbridge_library.capabilities.publish.Publish
platform                      | 2021-03-12 18:56:00+0000 [-]  - rosbridge_library.capabilities.subscribe.Subscribe
platform                      | 2021-03-12 18:56:00+0000 [-]  - <class 'rosbridge_library.capabilities.defragmentation.Defragment'>
platform                      | 2021-03-12 18:56:00+0000 [-]  - rosbridge_library.capabilities.advertise_service.AdvertiseService
platform                      | 2021-03-12 18:56:00+0000 [-]  - rosbridge_library.capabilities.service_response.ServiceResponse
platform                      | 2021-03-12 18:56:00+0000 [-]  - rosbridge_library.capabilities.unadvertise_service.UnadvertiseService
platform                      | 2021-03-12 18:56:00+0000 [-] WebSocketServerFactory starting on 9090
platform                      | 2021-03-12 18:56:00+0000 [-] Starting factory <autobahn.twisted.websocket.WebSocketServerFactory object at 0x7f16a58f2110>
platform                      | 2021-03-12 18:56:00+0000 [-] 1615575360.407555 | INFO | <module>:312 | Rosbridge WebSocket server started at ws://0.0.0.0:9090
platform                      | 2021-03-12 19:01:59+0000 [autobahn.twisted.websocket.WebSocketServerFactory] (TCP Port 9090 Closed)
platform                      | 2021-03-12 19:01:59+0000 [-] Stopping factory <autobahn.twisted.websocket.WebSocketServerFactory object at 0x7f16a58f2110>
platform                      | 2021-03-12 19:01:59+0000 [-] Main loop terminated.
platform                      | 1615575360.431470 | INFO | <module>:202 | Rosapi started
platform                      | Unhandled exception in thread started by 
platform                      | sys.excepthook is missing
platform                      | lost sys.stderr
roscore                       | Checking log directory for disk usage. This may take awhile.
roscore                       | Press Ctrl-C to interrupt
roscore                       | Done checking log file disk usage. Usage is <1GB.
roscore                       | 
roscore                       | started roslaunch server http://127.0.0.1:45603/
roscore                       | ros_comm version 1.12.17
roscore                       | 
roscore                       | 
roscore                       | SUMMARY
roscore                       | ========
roscore                       | 
roscore                       | PARAMETERS
roscore                       |  * /rosdistro: kinetic
roscore                       |  * /rosversion: 1.12.17
roscore                       | 
[... 114 lines cut ...]
carma-mock-comms-driver exited with code 137
carma-mock-lightbar-driver exited with code 137
carma-srx-controller-driver exited with code 137
carma-mock-lidar-driver exited with code 137
carma-mock-imu-driver exited with code 137
carma-mock-roadway-sensor-driver exited with code 137
`

### Carma version where this issue was discovered

CARMA3
Ubuntu 16.04
Docker version 20.10.5
### Expected behavior
CARMA should run successfully.

### Actual behavior

### Steps to reproduce the actual behavior

1. Run `carma start all`
2. Open localhost/index.html
3. Click on Start Platform button

### Related work

DokurOmkar, 2021-03-12:
> ### Types of Issue
> * [x]  Anomaly report (something appears to not work correctly)
> * [ ]  Enhancement request (describe the enhancement being requested)
> * [ ]  Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)
> 
> ### Descriptive summary
> I have set up a native Ubuntu 16.04 machine with the following configuration (8 core processor, 24GB RAM, GeForce 1650 Graphics) and followed the confluence page [here](https://usdot-carma.atlassian.net/wiki/spaces/CRMPLT/pages/470286338/Setup+a+VM+Ware+Image+for+Development) to set up the username, password, etc.
> 
> [image]
> 
> And I followed the entire documentation to set up the CARMA platform and then I am seeing several warnings after running `carma start all`
> 
> The warnings are:
> 
> [image]
> 
> `WARNING: Error loading config file: /home/carma/.docker/config.json: open /home/carma/.docker/config.json: permission denied`
> 
> [image]
> 
> `roscore | WARNING: Could not change permissions for folder [/opt/carma/logs/8463e2a6-8364-11eb-aa37-5cf9dd6b91fb], make sure that the parent folder has correct permissions.`
> 
> [image]
> 
> `carma-mock-lightbar-driver | 1615575336.685362033 | FATAL | main:188 | Error opening file: /opt/carma/data/mock_drivers/drivers.bag`
> 
> [image]
> 
> `platform | process[environment/lanelet2_map_visualization-19]: started with pid [215ERROR: cannot launch node of type [motion_computation/motion_computation_node]: can't locate node [motion_computation_node] in package [motion_computation]`
> 
> With this, I have opened the localhost/index.html. When I clicked on Start Platform button, I am seeing errors in the terminal which is below:
> 
> [image]
> [image]
> [image]
> 
> ### Carma version where this issue was discovered
> CARMA3
> Ubuntu 16.04
> Docker version 20.10.5
> 
> ### Expected behavior
> CARMA should run successfully.
> 
> ### Actual behavior
> Please see the screenshot below
> 
> [image]
> 
> ### Steps to reproduce the actual behavior
> 1. Run `carma start all`
> 2. Open localhost/index.html
> 3. Click on Start Platform button
> 
> ### Related work

kjrush, 2021-03-15:
Thanks for posting this! I'll dig a little bit deeper into it, but at first glance it looks like there may be something misconfigured with your `/opt/carma` folder permissions,  as it fails to find or access the bag file required for operating the mock drivers nodes. That may explain the errors your seeing as it would cause the system to shutdown after some period of not communicating with the mock drivers, but there may also be multiple factors at play.

DokurOmkar, 2021-03-15:
> Thanks for posting this! I'll dig a little bit deeper into it, but at first glance it looks like there may be something misconfigured with your `/opt/carma` folder permissions, as it fails to find [...]

Sure, please let me know if you need anything else to replicate the issue on your end. The only thing which differs from the Confluence Developer User Guide is that I am using a  dedicated machine running Ubuntu 16.04 and not the one with VMWare.

Fy2333, 2021-03-17:
Hi Omkar, 

Hope I can provide some help here. This problem may be related to your SSH keys. Please follow [this link](https://docs.github.com/en/github/authenticating-to-github/testing-your-ssh-connection) to check your SSH keys, if you receive a "permission denied" message, follow the instruction to fix it.
Then try to re-cloning the repository (clean previous file if needed)

Zhaohui Liang
University of South Florida 
[CATS lab](http://cee.eng.usf.edu/faculty/xiaopengli/CATS%20Lab.html)

DokurOmkar, 2021-03-17:
> Hi Omkar,
>
> Hope I can provide some help here. This problem may be related to your SSH keys. Please follow [this [...]

Hi Liang, I have followed the link provided by you and I can see that I am successfully been authenticated as shown in the screenshot below:

[image]

Please advise if this looks ok! Thank you!

msmcconnell, 2021-03-22:
@DokurOmkar What is the exact version you are running (3.4.3, develop, 3.5.0 etc.) ?

One potential issue is that you are launching via the terminal while also trying to launch via the UI. 
Give this a shot
1. carma start all
2. Open UI in web browser
3. uncheck "Remote Launch" if on 3.4 and earlier otherwise Check "Debug Mode" if on develop or 3.5
4. Click Ok/Start

If the remote launch or debug mode is not marked, the UI will try to launch the docker containers directly which might be clashing with your terminal launch. The other errors you are seeing on the terminal shouldn't be causing the system to crash, so lets focus on keeping carma running first.

DokurOmkar, 2021-03-23:
> @DokurOmkar What is the exact version you are running (3.4.3, develop, 3.5.0 etc.) ?
>
> One potential issue is that you are launching via the terminal while also trying to launch via the UI.
> Give this [...]

Hi @msmcconnell, After looking at the version in docker-compose.yml, I see that I am using `usdotfhwastoldev/carma-base:develop`. Please see the screenshot attached below:

[image]

After seeing that I have a 'develop' image, I have followed the steps given by you and did steps 1,2 and 3 and came at the following point as shown in the below image:

[image]

I clicked on Start Platform and the 'Route Page' appeared. I clicked on 'Town_02_laneet111_route' and I am getting the following alert on the UI:
[image]

I tried on clicking the 'Restart' Button and I am getting the following after that:
[image]

And clicking on 'Logout' is stopping the 'carma start all' on the terminal.

Please advise on how to fix this.

msmcconnell, 2021-03-23:
@DokurOmkar At this point, what you are seeing is consistent with a local installation that does not have a rosbag to playback. So what you have shared demonstrates a running build/installation. 
The next step would be to determine your immediate needs. Do you have a vehicle you can try installing the system onto or are you limited to simulation?

DokurOmkar, 2021-03-23:
> @DokurOmkar At this point, what you are seeing is consistent with a local installation that does not have a rosbag to playback. So what you have shared demonstrates a valid build/installation.
> The next step would be to determine your immediate needs. Do you have a vehicle you can try installing the system onto or are you limited to simulation?

@msmcconnell Thank you for confirming that I have a valid build/installation. At the moment, we are only trying to run CARMA in a simulated environment. Please advise on what are the next steps for running CARMA in simulation. Is there a confluence page where I can follow?

Thank you!

DokurOmkar, 2021-03-23:
> @DokurOmkar At this point, what you are seeing is consistent with a local installation that does not have a rosbag to playback. So what you have shared demonstrates a valid build/installation.
> The [...]

@msmcconnell Also, how can we add a recorded rosbag file to the CARMA platform. We have several recorded rosbag files within our university. Please advise how we can playback those files using the CARMA platform.

Thank you!

msmcconnell, 2021-03-23:
@DokurOmkar A recorded rosbag can be placed in /opt/carma/data/mock_drivers/drivers.bag and the system will attempt to playback the raw sensor data from this bag. It requires topics adhere to the naming conventions for drivers described in section
3.2.1.6 here of the architecture here: https://usdot-carma.atlassian.net/wiki/spaces/CRMPLT/pages/89587713/CARMA+Platform+System+Architecture?preview=/89587713/1324384490/CARMA%20Platform%20Architecture%20-%20ROS2.docx

Since architectures are sometimes out of date you can also look at the subscriptions for each of these files https://github.com/usdot-fhwa-stol/carma-platform/tree/develop/mock_drivers/rosbag_mock_drivers/src

You need the Controller, GPS, and Lidar drivers at a minimum. Having the ROS bag would allow you to run localization in simulation and select a route (if you have an HD map as well), but it will **not** allow you to simulate motion since there is no vehicle feedback. 

For full vehicle simulation you are better off using the CARLA simulator. An adapter for CARLA and CARMA can be found here
https://github.com/usdot-fhwa-stol/carma-simulation
Setup instructions https://usdot-carma.atlassian.net/wiki/spaces/CRMPLT/pages/848166928/CARMA+Execution+Environment+Simulation+Configuration

@kjrush is the best person to follow up with regarding further questions on simulation.

DokurOmkar, 2021-03-23:
> @DokurOmkar A recorded rosbag can be placed in /opt/carma/data/mock_drivers/drivers.bag and the system will attempt to playback the raw sensor data from this bag. It requires topics adhere to the [...]

@msmcconnell Thank you for all the information. I see that there is no rosbag file on my local at the location you mentioned and that's the reason it's giving me a system when trying to start a route. I will try the adapter for CARLA and CARMA provided by you at https://github.com/usdot-fhwa-stol/carma-simulation. 

Also, I am not able to access this link on setting up the simulation environment https://usdot-carma.atlassian.net/wiki/spaces/CRMPLT/pages/848166928/CARMA+Execution+Environment+Simulation+Configuration

Can you please provide access to my Atlassian account (omkardokur@mail.usf.edu) so that I can set up the simulator environment on my local?

Thank you!

msmcconnell, 2021-03-23:
@DokurOmkar you should be able to see the doc now.

DokurOmkar, 2021-03-23:
> @DokurOmkar you should be able to see the doc now.

Thank you @msmcconnell, I can see the document now.

DokurOmkar, 2021-03-24:
> @DokurOmkar you should be able to see the doc now.

@kjrush I am encountering a problem with NumPy while following the confluence page at https://usdot-carma.atlassian.net/wiki/spaces/CRMPLT/pages/848166928/CARMA+Execution+Environment+Simulation+Configuration to set up carma-carla simulation.

I am getting the following error when trying to install NumPy:

[image]

May I please know if you have encountered the same issue too? Please advise on how to fix this issue!

Thank you!

DokurOmkar, 2021-03-30:
> @DokurOmkar you should be able to see the doc now.
@msmcconnell @kjrush May I please know which version of python is required here? Currently, I have Python 2.7.12 and 
Python 3.5.2 installed on my machine. Should I upgrade it to Python 3.7 instead?

Please advise!!

Thank you!

DokurOmkar, 2021-03-30:
@msmcconnell @kjrush I was able to fix the above error by following this: https://stackoverflow.com/questions/41773540/how-to-install-numpy-to-python-3-5/41773560

Now, I have installed NumPy for python3.5 and able to run the command `python3.5 CARLA_server_interface.py` without any errors as shown below:

[image]

Thank you for providing the documentation again!

closed 2021-03-30 by DokurOmkar