usdot-fhwa-stol/carma-platform #1212: Black Pacifica NAS is not connecting

MishkaMN, 2021-04-21:
### Types of Issue

- [X] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

NAS Windows computer no longer has the CarmaBPacifica username, which is why it is not able to connect.
That username and WindowsSharedFolder on the network should be created inside to properly connect.

### Carma version where this issue was discovered

carma-system 3.5.3+
### Expected behavior
Spectra computer is able to connect to NAS using the username and password configured in its Ubuntu's /etc/rc.local file
### Actual behavior

rc.local fails on Ubuntu
investigation showed that there was no CarmaBPacfica user
### Steps to reproduce the actual behavior

### Related work

MishkaMN, 2021-04-21:
This username is on the I2 network now. So the username is `I2-CarmaBlkPac` with same password. Need verification

msmcconnell, 2021-05-26:
Verified as part of zephyr verification testing. Closing issue.

closed 2021-05-26 by msmcconnell