usdot-fhwa-stol/carma-platform #1214: Geofence request message is not being broadcast by the OBU because of wrong PSID

snallamothu, 2021-04-21:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
Geofence request is not being broadcast by the OBU because PSID being used (244) is not recognized by the OBU.  PSID 8003 should be used.  

### Carma version where this issue was discovered
carma-system 3.5.3+

### Expected behavior
The message should be broadcast by the OBU for the RSU to forward it to V2X-Hub in order to get a response from CARMA Cloud.

### Actual behavior
V2X-Hub doesn't receive the request.

### Steps to reproduce the actual behavior
Start CARMA UI and select a route.

### Related work

snallamothu, 2021-04-21:
[image]

msmcconnell, 2021-05-26:
This has been resolved

closed 2021-05-26 by msmcconnell