usdot-fhwa-stol/carma-platform #1312: Blue Lexus NAS shuts down and restarts occasionally

MishkaMN, 2021-06-15:
### Types of Issue

- [ ] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [X] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

Blue Lexus's NAS shuts down itself and restarts again sometimes. It is possible that it is power issue. This could affect recodings of rosbag if happened during a run. It restarts itself and NAS connects back autoamtically in 5-7 mins. It doesn't turn off that often, but noticed 5-7 times for a whole day of testing.

### Carma version where this issue was discovered

release/zephyr
### Expected behavior
See above.
### Actual behavior

See above
### Steps to reproduce the actual behavior

### Related work

msmcconnell, 2021-06-21:
Unclear if the same root cause, but today (6/21). I encountered numerous power issues with the blue lexus PC and screens shutting off randomly during the day.

msmcconnell, 2021-07-27:
The blue lexus battery was identified as insufficient for our use case and replaced with a battery intended for longer use. I am going to close this issue for now as the only known cause has been resolved

closed 2021-07-27 by msmcconnell