usdot-fhwa-stol/carma-platform #1318: Colcon Fails to Build Packages

GoodarzMehr, 2021-06-23:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
Based on the instructions [here](https://usdot-carma.atlassian.net/wiki/spaces/CRMPLT/pages/491651073/Using+a+CARMA+Platform+development+environment), my understanding is that to modify the mock camera driver CARMA needs to be rebuilt. However, colcon fails to build the required packages.

### Carma version where this issue was discovered
3.6.0

### Expected behavior
CARMA packages built using colcon.

### Actual behavior
Colcon fails to build colcon packages. Here's the terminal output:

```
Attempting to build CARMA
CARMA Workspace: /workspaces/carma_ws/carma
Autoware Source Dir: /workspaces/carma_ws/autoware.ai

Skipping Autoware build due to -x option
Building CARMA
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3.5/distutils/core.py", line 216, in run_setup
    exec(f.read(), g, l)
  File "<string>", line 8
    with open(f"{setup_dir}/README.md", "r") as fh:
                                     ^
SyntaxError: invalid syntax
[0.567s] ERROR:colcon.colcon_core.package_identification:Exception in package identification extension 'python_setup_py' in 'src/opendrive2lanelet': Command '['/usr/bin/python3', '-c', "import sys;from setuptools.extern.packaging.specifiers import SpecifierSet;from distutils.core import run_setup;dist = run_setup(    'setup.py', script_args=('--dry-run',), stop_after='config');skip_keys = ('cmdclass', 'distclass', 'ext_modules', 'metadata');data = {    key: value for key, value in dist.__dict__.items()     if (        not key.startswith('_') and         not callable(value) and         key not in skip_keys and         key not in dist.display_option_names    )};data['metadata'] = {    k: v for k, v in dist.metadata.__dict__.items()     if k not in ('license_files', 'provides_extras')};sys.stdout.buffer.write(repr(data).encode('utf-8'))"]' returned non-zero exit status 1
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/colcon_core/package_identification/__init__.py", line 143, in _identify
    retval = extension.identify(_reused_descriptor_instance)
  File "/usr/lib/python3/dist-packages/colcon_python_setup_py/package_identification/python_setup_py.py", line 48, in identify
    config = get_setup_information(setup_py)
  File "/usr/lib/python3/dist-packages/colcon_python_setup_py/package_identification/python_setup_py.py", line 242, in get_setup_information
    setup_py, env=env)
  File "/usr/lib/python3/dist-packages/colcon_python_setup_py/package_identification/python_setup_py.py", line 283, in _get_setup_information
    cwd=os.path.abspath(str(setup_py.parent)), check=True, env=env)
  File "/usr/lib/python3.5/subprocess.py", line 708, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['/usr/bin/python3', '-c', "import sys;from setuptools.extern.packaging.specifiers import SpecifierSet;from distutils.core import run_setup;dist = run_setup(    'setup.py', script_args=('--dry-run',), stop_after='config');skip_keys = ('cmdclass', 'distclass', 'ext_modules', 'metadata');data = {    key: value for key, value in dist.__dict__.items()     if (        not key.startswith('_') and         not callable(value) and         key not in skip_keys and         key not in dist.display_option_names    )};data['metadata'] = {    k: v for k, v in dist.metadata.__dict__.items()     if k not in ('license_files', 'provides_extras')};sys.stdout.buffer.write(repr(data).encode('utf-8'))"]' returned non-zero exit status 1

Starting >>> j2735_msgs
Starting >>> lib_vehicle_model
Starting >>> wgs84_utils
Starting >>> can_msgs
Starting >>> novatel_gps_msgs
Starting >>> asn1c
Starting >>> sigpack
Starting >>> socketcan_interface
Starting >>> ssc_ds_fusion
Starting >>> ssc_ne_pacifica
Starting >>> ssc_pm_lexus
Starting >>> uncertainty_tools
Starting >>> ssc_gear_change
Starting >>> ssc_pm_cascadia_dot_10002
Starting >>> ssc_pm_cascadia_dot_10003                                                                                                                                           
Starting >>> ssc_pm_cascadia_dot_10004
Starting >>> ssc_pm_cascadia_dot_80550
--- stderr: ssc_ds_fusion                                                                                                                                                       
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "dbw_mkz_msgs" with
  any of the following names:

    dbw_mkz_msgsConfig.cmake
    dbw_mkz_msgs-config.cmake

  Add the installation prefix of "dbw_mkz_msgs" to CMAKE_PREFIX_PATH or set
  "dbw_mkz_msgs_DIR" to a directory containing one of the above files.  If
  "dbw_mkz_msgs" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:22 (find_package)

CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "dbw_mkz_msgs" with
  any of the following names:

    dbw_mkz_msgsConfig.cmake
    dbw_mkz_msgs-config.cmake

  Add the installation prefix of "dbw_mkz_msgs" to CMAKE_PREFIX_PATH or set
  "dbw_mkz_msgs_DIR" to a directory containing one of the above files.  If
  "dbw_mkz_msgs" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:22 (find_package)

---
Failed   <<< ssc_ds_fusion [1.03s, exited with code 1]
Aborted  <<< socketcan_interface [1.05s]
Aborted  <<< asn1c [1.06s]
Aborted  <<< ssc_ne_pacifica [1.04s]
Aborted  <<< ssc_pm_lexus [1.03s]
Aborted  <<< lib_vehicle_model [1.09s]
Aborted  <<< uncertainty_tools [1.02s]
Aborted  <<< ssc_gear_change [1.02s]
Aborted  <<< novatel_gps_msgs [1.07s]                                                                                                                            
Aborted  <<< sigpack [1.06s]
Aborted  <<< ssc_pm_cascadia_dot_10003 [1.01s]
Aborted  <<< j2735_msgs [1.10s]
Aborted  <<< ssc_pm_cascadia_dot_10002 [1.01s]
Aborted  <<< ssc_pm_cascadia_dot_80550 [1.01s]
Aborted  <<< ssc_pm_cascadia_dot_10004 [1.02s]
Aborted  <<< can_msgs [1.11s]
Aborted  <<< wgs84_utils [1.17s]

Summary: 0 packages finished [2.05s]
  1 package failed: ssc_ds_fusion
  16 packages aborted: asn1c can_msgs j2735_msgs lib_vehicle_model novatel_gps_msgs sigpack socketcan_interface ssc_gear_change ssc_ne_pacifica ssc_pm_cascadia_dot_10002 ssc_pm_cascadia_dot_10003 ssc_pm_cascadia_dot_10004 ssc_pm_cascadia_dot_80550 ssc_pm_lexus uncertainty_tools wgs84_utils
  1 package had stderr output: ssc_ds_fusion
  69 packages not processed
CARMA built successfuly. Binaries sourced from /workspaces/carma_ws/carma/install/setup.bash
```

### Steps to reproduce the actual behavior

1. In the VS Code terminal run
```
cd /workspaces/carma_ws/carma/src/carma-platform
./carma_build.bash -c /workspaces/carma_ws/carma/ -x
```

### Related work

msmcconnell, 2022-01-12:
@GoodarzMehr sorry no-one followed up on this. Does this still pose an issue for you, or can I close this?

GoodarzMehr, 2022-01-13:
@msmcconnell thanks for reminding me about this. It has been resolved, so I'm closing this.

closed 2022-01-13 by GoodarzMehr