usdot-fhwa-stol/carma-platform #1331: Deployer Resources for Platforms with Different Hardware

GoodarzMehr, 2021-06-29:
### Types of Issue

- [ ] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [x] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
It would helpful to the broader community to have [deployer resources](https://usdot-carma.atlassian.net/wiki/spaces/CRMPLT/pages/1032061274/Deployer+Resources) available to enable successful deployment of CARMA on platforms that use different hardware. This would allow broader engagement with the CARMA Platform.

One beneficial resource would be general instructions on how to modify availabe ROS packages and build a docker image that acts as a driver for a particular sensor. For example, similar to what the developer team has done for Velodyne lidars, if a user is using Ouster lidars, the resources above can help the user modify [this](https://github.com/ouster-lidar/ouster_example) ROS package to act as a CARMA driver. Another example would modifying the [GSCam](https://github.com/ros-drivers/gscam) package to act as a driver for cameras that are accessed through a [GStreamer](https://gstreamer.freedesktop.org/) pipeline.

Deployer resources could also provide instructions on modifying a Dockerfile to build a Docker image of a particular driver and add that driver to the `docker-compose.yml` file. They could also address how to handle situations where two or more sensors of the same type are present. For example, if three cameras are present, should they all publish to the same `/hardware_interface/camera/image_raw` topic?

### Carma version where this issue was discovered
3.5.3

### Expected behavior
N/A

### Actual behavior
N/A

### Steps to reproduce the actual behavior
N/A

### Related work
N/A

msmcconnell, 2021-06-30:
Thanks for bring this up. We wholeheartedly agree with you that more documentation in this area is needed, and we have it on our backlog. Unfortunately, I cannot provide a timeline for its release yet, as the core development team has a number of near term deadlines that are being given high priority. If this is a blocker for your own work, I would recommend also emailing CARMA@dot.gov about this to see if the priority can be increased.

GoodarzMehr, 2021-07-01:
Thanks for letting me know. We would certainly appreciate the resources, as our sensory platform is quite different from the FHWA one. In the meantime, we will try and see if we can create drivers for our sensors, and if successful will share them online for others to use as well.

One quick question: since we are using a Ford Fusion with the Dataspeed Drive-X-Wire controller which is similar to what you are using, do we just connect to its usb port and launch the proper controller image (the one in the ford_fusion_sehybrid_2019 config) or are there other steps we need to take?

msmcconnell, 2021-07-04:
@GoodarzMehr Configs apply to classes of vehicles so in this sense you are correct. However, along with the config you will need a vehicle calibration folder which contains all the individual vehicle configurations. 
See here https://github.com/usdot-fhwa-stol/carma-config#example-opt-folder
and 
https://github.com/usdot-fhwa-stol/carma-config/tree/develop/example_calibration_folder
For examples