usdot-fhwa-stol/carma-platform #1356: Routing Graph creation takes unacceptably long time on ACM map

msmcconnell, 2021-07-20:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

While testing Basic Travel use case at ACM it was observed that the vehicle appeared to lag heavily when the geofence message was received. Analysis of the log files indicated that this was due to the routing graph build triggered by map updates. This gave the following behavior:

- In carma_wm_ctrl. It takes 27 seconds for the geofence to be processed.
- In carma_wm, it takes 67 seconds for the update to be applied. These are really extreme times.

The reason this did not happen previously is not fully known, however, initial indications suggest it is due to map size. The ACM map is 6 times larger (number of lines) than the Summit Point map. This is substantially more geometry to be analyzed by the Lanelet2 system. 
### Carma version where this issue was discovered

develop after 3.6
### Expected behavior
Basic travel can be completed when the speed advisory geofence is received. 
### Actual behavior

See description
### Steps to reproduce the actual behavior

Run the test/platoon6 branch at ACM with carma-cloud setup to broadcast a speed advisory. 

### Related work

JonSmet, 2022-04-14:
This issue was encountered at Suntrax when testing the Freight Workzone use case as well (when TCMs were received by carma-platform and processed/applied to the map, the system began to lag and the vehicle drifted out of its lane). From logs, the routing graph took approximately 0.3-0.6 seconds to be re-computed. It is believed that since this re-computation occurs in several plugins within the vehicle control loop, it is the main contributor to the vehicle performance issue. This behavior was repeatable.

msmcconnell, 2022-04-20:
The cost of re-computing the graph was limited to a single node by the above PRs. After this change, some additional reduction in excess logging allowed for the system to operate normally. I will close this issue for now.

closed 2022-04-20 by msmcconnell