usdot-fhwa-stol/carma-platform #1436: Mock Controller Driver Dropping Connection Due to Message md5sum Mismatch

GoodarzMehr, 2021-09-14:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

When running CARMA with all the Mock Drivers, the following error message is displayed during the startup:
`ERROR | Publication::validateHeader:502 | Client [/hardware_interface/controller] wants topic /hardware_interface/vehicle_cmd to have datatype/md5sum [autoware_msgs/VehicleCmd/153301368e83526f24c12f8e8cddf283], but our version has [autoware_msgs/VehicleCmd/8e7481711780182446911d94fa3a384d]. Dropping connection.`

### Carma version where this issue was discovered

We first saw this behavior after updating to version 3.5.0, and the error has persisted even after updating to the latest version (3.7.1).

### Expected behavior
Maintained connection between client `/hardware_interface/controller` and topic `/hardware_interface/vehicle_cmd`.

### Actual behavior

Dropped connection between client `/hardware_interface/controller` and topic `/hardware_interface/vehicle_cmd`.

### Steps to reproduce the actual behavior

1. Run `carma start all roscore platform mock-camera-driver mock-lidar-driver mock-can-driver mock-controller-driver mock-gnss-driver`

### Related work

msmcconnell, 2021-09-15:
Thanks for posting this. I believe the 3.7.1 release had a mismatched docker base image used and the 3.7.2 image should resolve this issue. 

@kjrush can provide additional details.

GoodarzMehr, 2021-10-06:
Update: This issue does not seem to have been resolved in the newer versions.

We synchronized VSCode with the develop branch of `carma-base`, `autoware.ai`, and `carma-platform`, removed all our Docker images, and rebuilt them in that order. We still saw that same error message when running `carma start all roscore platform mock-camera-driver mock-lidar-driver mock-can-driver mock-controller-driver mock-gnss-driver`.

The issue seems to stem from the fact that the `autoware_msgs` package in `carma-base` is installed by `apt` from `ros-kinetic-autoware-msgs`, where VehicleCmd uses the older `int32 gear` field, whereas in the `carma-platform` the package is built from source and VehicleCmd uses the newer `autoware_msgs/Gear gear_cmd` field. Perhaps changing the Dockerfile of `carma-base` to build `autoware_msgs` from source can fix this issue.

msmcconnell, 2021-11-08:
This should be resolved on the develop branch now by the linked PR. Closing

closed 2021-11-08 by msmcconnell