usdot-fhwa-stol/carma-platform #1442: GStreamer Camera Driver for CARMA

GoodarzMehr, 2021-09-18:
### Types of Issue

- [ ] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [x] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

We have developed a GStreamer camera driver for CARMA that is available at [this](https://github.com/VT-ASIM-LAB/gstreamer_camera_driver/) repository. The driver is based on the ROS GSCam package that publishes images to a ROS topic from a GStreamer camera stream. The Code has been modified to ensure it satisfies CARMA's requirements. Namely, that it publishes images in BGR8 encoding and that it also publishes a driver discovery message at 1.25 Hz. We would love to hear your feedback about the driver, and if you find it useful, feel free to include it on the main CARMA repository.

### Carma version where this issue was discovered

N/A

### Expected behavior
N/A

### Actual behavior

N/A

### Steps to reproduce the actual behavior

N/A

### Related work

N/A

msmcconnell, 2021-09-22:
This is great to see @GoodarzMehr ! We will definitely take a look an probably add a link to the repo in one of our documents.

msmcconnell, 2021-09-28:
I took a quick look and it seems the driver is setup correctly to interface with carma-platform.

GoodarzMehr, 2021-10-01:
It's great to hear that! We have also created hardware drivers for Ouster lidars and Inertial Labs GNSS products that we will be uploading soon, which should hopefully expand the hardware compatibility of the CARMA platform. We are also working on a Dataspeed interface driver that uses Dataspeed's own ULC controller which should remove the need for a separate proprietary software like AutonomouStuff's SSC.

closed 2021-10-01 by GoodarzMehr