usdot-fhwa-stol/carma-platform #1444: a1 release verification testing fixes

MishkaMN, 2021-09-20:
### Types of Issue

- [X] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

Verification testing needs to be done on workzone testing

### Carma version where this issue was discovered

carma 3.7.2 + 
### Expected behavior
NA
### Actual behavior

NA
### Steps to reproduce the actual behavior

1. Do this
1. Then do this...

### Related work

msmcconnell, 2021-09-22:
This issue seems to be a catch all for verification testing issues. It is better to reference specific issues than a generic "is issue". We will leave open until all issues related to verification testing are logged.

msmcconnell, 2021-09-24:
Release has been cut as 3.8.0 and this issue can be closed.

closed 2021-09-24 by msmcconnell