usdot-fhwa-stol/carma-platform #1456: Guidance/Platoon_Strategic throws a Shutdown error

lewisid, 2021-09-22:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

When testing the Work Zone Use Case functionality on the Black Pacifica, the vehicle stopped at the red traffic signal and began moving forward.  Once it turned green , it sent a SHUTDOWN message stating health_monitor requesting shutdown due to: Uncaught Exception in /guidance/platoon_strategic exception: Nullptr passed to constructor
### Carma version where this issue was discovered

### Expected behavior
The vehicle should be able to traverse the route without any errors.

### Actual behavior

See description

### Steps to reproduce the actual behavior

1. Position the rear axel of the vehicle aligned with the metal truss across innovation drive
2. Localize and select the route
3. Wait for a full cycle of the light (red,green,yellow) before engaging.
4. Engage. For the red stopping case, I suggest engaging on Green with 5s left which you can see on the UI
5. For the green case, engage on Red 15.

### Related work

msmcconnell, 2021-09-22:
Might be related to https://github.com/usdot-fhwa-stol/carma-platform/issues/1451

msmcconnell, 2021-09-23:
Testing suggests this was resolve by the above PR

closed 2021-09-23 by msmcconnell