usdot-fhwa-stol/carma-platform #1519: MobilityPath and MobilityOperation header hostBSMId  are incorrect

dan-du-car, 2021-11-03:
### Types of Issue
Both MobilityPath and MobilityOperation header hostBSMId  are incorrect. The BSM id total length is 8, however both MobilityPath and MobilityOperation are sending 10 digits BSM id to cpp_message node. So cpp_message node will ignore this 10 digits bsm id and replace it with 8 digits default id = "00000000". 

- [ ] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

 
-- This part should belong to the regional extension frame for --
-- the usage of mobility message. Due to the limitation of ASN1C, --
-- we put them in here for the temporary use. --
BSM id J2735 field type:

MobilityDynamicID ::= IA5String (SIZE(8))
   -- mobility dynamic ID is the same as host vehicle current BSM ID in hex string

### Carma version where this issue was discovered

### Expected behavior

### Actual behavior

### Steps to reproduce the actual behavior

1. Do this
1. Then do this...

### Related work

dan-du-car, 2021-11-08:
Please double check the J2735 BSM message VehicleSize.length value is converted to centimeters.
https://github.com/usdot-fhwa-stol/carma-messenger/blob/564833996efad24056cc1836a101640a84791d3b/carma-messenger-core/cpp_message/src/BSM_Message.cpp#L171

msmcconnell, 2021-12-22:
This was resolved by above PRs

closed 2021-12-22 by msmcconnell