usdot-fhwa-stol/carma-platform #1568: Unable to Build Autoware.ai from Source

GoodarzMehr, 2021-12-21:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

When building the Autoware.ai Docker image from source, package autoware_camera_lidar_calibrator fails to build, and so does the Docker image.

### Carma version where this issue was discovered

3.10.0

### Expected behavior
Autoware.ai Docker image built from source successfully.

### Actual behavior

The build process outputs the following:
```
Starting >>> autoware_camera_lidar_calibrator
--- stderr: autoware_camera_lidar_calibrator
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** io features related to libusb-1.0 will be disabled
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: option --install-layout not recognized
CMake Error at catkin_generated/safe_execute_install.cmake:4 (message):
  
  execute_process(/home/carma/autoware.ai/build/autoware_camera_lidar_calibrator/catkin_generated/python_distutils_install.sh)
  returned error code
Call Stack (most recent call first):
  cmake_install.cmake:56 (include)

---
Failed   <<< autoware_camera_lidar_calibrator [9.97s, exited with code 1]
```

### Steps to reproduce the actual behavior

1. `cd ~/carma_ws/src/autoware.ai/docker`
2. `sudo ./build-image.sh -d`

### Related work

I have found that adding `-DSETUPTOOLS_DEB_LAYOUT=OFF` to the end of the `docker/install.sh` file can solve the issue, but this may be a temporary fix.

msmcconnell, 2021-12-21:
@GoodarzMehr Can you confirm what branch/tag you are building from?

GoodarzMehr, 2021-12-21:
@msmcconnell We are on the `develop` branch of `carma-platform` and `carma-develop` branch of `autoware.ai`. I last synced our local repos with GitHub on 12/20/2021, before attempting to build them.

msmcconnell, 2021-12-22:
After fixing a small issue in carma-base, I was able to replicate this error.
```
docker pull usdotfhwastoldev/carma-base:develop
cd ~/carma_ws/src/autoware.ai/docker
git checkout carma-develop
git pull
./build-image.sh -d
```

Gives the following exception which matches what was reported above.
```
Finished <<< astar_search [19.5s]
Starting >>> autoware_camera_lidar_calibrator
--- stderr: autoware_camera_lidar_calibrator
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** io features related to libusb-1.0 will be disabled
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: option --install-layout not recognized
CMake Error at catkin_generated/safe_execute_install.cmake:4 (message):
  
  execute_process(/home/carma/autoware.ai/build/autoware_camera_lidar_calibrator/catkin_generated/python_distutils_install.sh)
  returned error code
Call Stack (most recent call first):
  cmake_install.cmake:56 (include)

---
Failed   <<< autoware_camera_lidar_calibrator [15.7s, exited with code 1]

Summary: 24 packages finished [8min 21s]
  1 package failed: autoware_camera_lidar_calibrator
  9 packages had stderr output: astar_search autoware_camera_lidar_calibrator gnss ndt_cpu ndt_tku pcl_omp_registration qpoases_vendor ros_observer system_monitor
  47 packages not processed
The command '/bin/sh -c ./home/carma/autoware.ai/docker/install.sh' returned a non-zero code: 1
Error response from daemon: No such image: usdotfhwastoldev/autoware.ai:develop
Tagged usdotfhwastoldev/autoware.ai:develop as usdotfhwastoldev/autoware.ai:latest
```
Further investigation identified this thread https://github.com/ros/catkin/issues/863 which discusses how the 
```-DSETUPTOOLS_DEB_LAYOUT=OFF``` option can be used as a work around. The root cause appears to be a non-portable call in catkin's python build that depends on setuptools. I'm not overly familiar with python installations and the impact of this argument. I think further input may be needed from the team. If we can safely add this option, then 
https://github.com/usdot-fhwa-stol/autoware.ai/blob/5f658d40d731c97ea780afd9986a1cb85a9e2bd1/docker/install.sh#L28 could be modified to be to include this with the other ```--cmake-args```

msmcconnell, 2021-12-22:
For development team, this is the corresponding Jira ticket
https://usdot-carma.atlassian.net/browse/CAR-5122

GoodarzMehr, 2021-12-22:
@msmcconnell I just want to note that the a similar thing also happens when trying to build the Docker image for `carma-platform`. A similar error pops up when trying to build the `carma` package.

adev4a, 2022-01-06:
I believe this issue has been fixed with https://github.com/usdot-fhwa-stol/carma-base/pull/111. Autoware.ai and carma-platform builds were verified. 
@GoodarzMehr you'll need to update your carma-base image first for the fix to work.

GoodarzMehr, 2022-01-06:
I was able to build both autoware.ai and carma-platform from source and can confirm that the issue has been resolved.

closed 2022-01-06 by GoodarzMehr