usdot-fhwa-stol/carma-platform #1574: Additional Hardware Drivers for CARMA

GoodarzMehr, 2022-01-03:
### Types of Issue

- [ ] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [x] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

We have recently published the hardware drivers that we developed as part of implementing the CARMA Platform on our vehicle. Since we use sensors and hardware that are different (and much less expensive) that the ones officially supported, we hope this would help expand the compatibility of the CARMA Platform and facilitate its adoption. The hardware drivers that we have developed are as follows:

[GStreamer Camera Driver](https://github.com/VT-ASIM-LAB/gstreamer_camera_driver): this driver can be used for broadcasting any [GStreamer](http://gstreamer.freedesktop.org/)-based video stream. It generally supports any camera that has a V4L2 driver, in particular specialized automotive cameras from [Leopard Imaging](https://www.leopardimaging.com/), [E-con Systems](https://www.e-consystems.com/), etc.

[Ouster Lidar Driver](https://github.com/VT-ASIM-LAB/ouster_lidar_driver): this driver can be used for connecting to and configuring [Ouster](https://ouster.com/) sensors that can be a more affordable alternative to Velodyne lidars that are currently supported.

[Inertial Labs GNSS/INS Driver](https://github.com/VT-ASIM-LAB/inertiallabs_gnss_driver): this driver can be used for connecting to [Inertial Labs](https://inertiallabs.com/products/) GNSS/INS, IMU-P, AHRS, and AHRS-10 products that can be a more affordable alternative to NovAtel products that are currently supported.

Dataspeed Drive-by-Wire Kit [CAN](https://github.com/VT-ASIM-LAB/dataspeed_can_driver) and [ULC (Universal Lat/Lon Controller)](https://github.com/VT-ASIM-LAB/dataspeed_controller_driver) Drivers: these two drivers can be used on vehicles that are equipped with the Dataspeed Drive-by-Wire Kit to read CAN messages and control the vehicle through Dataspeed's built-in Universal Lat/Lon Controller (ULC). This should remove the need for the AutonomouStuff SSC software that is proprietary and essentially identical to the ULC.

We successfully implemented the CARMA Platform using these drivers on our 2017 Ford Fusion Hybrid SE with the following sensors:
- Ouster OS1-64U lidar
- Leopard Imaging LI-IMX390-GMSL2 cameras (connected via ConnectTech's GMSL2 camera board mounted on a Nvidia Jetson Xavier DevKit)
- Inertial Labs INS-D 
- Dataspeed Driver-by-Wire Kit

### Carma version where this issue was discovered

N/A

### Expected behavior
N/A

### Actual behavior

N/A

### Steps to reproduce the actual behavior

N/A

### Related work

N/A

msmcconnell, 2022-01-04:
Thanks for the awesome contribution @GoodarzMehr ! To help other users find these drivers I have put up a PR to our readme (linked above) to include links to these drivers. If you don't approve of that, please let me know.

GoodarzMehr, 2022-01-04:
Thanks @msmcconnell, this is great!

closed 2022-01-04 by GoodarzMehr