usdot-fhwa-stol/carma-platform #1634: Route is not selectable due to vehicle position not being set

MishkaMN, 2022-02-22:
### Types of Issue

- [X] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

When selecting route, CARMA is not able to select route because "vehicle position was not set". This is due to the `bumper_pose_cb` was never called. 
https://github.com/usdot-fhwa-stol/carma-platform/blob/develop/route/src/route_generator_worker.cpp#L478

It could be simple as calling that function from the node in `initialize()`, but there could be further issue behind this until fully tested. 

### Carma version where this issue was discovered

carma-system-3.11.0+ (develop)
### Expected behavior
Route selectable

### Actual behavior

Route is not selectable

### Steps to reproduce the actual behavior

Build and start carma from develop

### Related work

msmcconnell, 2022-02-22:
@JonSmet didn't you mention there was a fix for this on the way?

JonSmet, 2022-02-22:
@msmcconnell I was building a potential fix at the end of the day yesterday, but unfortunately it was not successful. I have another potential fix that I am going to test this morning and will report back.

msmcconnell, 2022-04-01:
Resolved by above PR

closed 2022-04-01 by msmcconnell